lanrat/homeplate #2: Stuck on input monitoring

C9H13NO3-dev, 2022-03-24:
Hey lanrat,

I managed to flash your homeplate software to my Inkplate, but after configuring it, it's stuck at "[INPUT] Starting input monitoring..." in the serial console - checking in my router, I don't see a wifi connection happening. This is the full boot log:

`20:14:57.021 > [SETUP] Battery: 85% (4.54v)
20:14:57.021 > [SETUP] starting button task
20:14:57.021 > [SETUP] starting time task
20:14:57.021 > [TIME] RTC local time (946687792) 1999-12-31 16:49
20:14:57.021 > [SETUP] starting WiFi task
20:14:57.021 > [SETUP] starting OTA task
20:14:57.021 > [SETUP] starting MQTT task
20:14:57.021 > [MQTT] starting MQTT
20:14:57.021 > [SETUP] starting sleep task
20:14:57.021 > [SETUP] starting activities task
20:14:59.848 > [INPUT] Starting input monitoring...`

I already double checked my wifi-credentials but that is perfectly fine...

Thanks in advance,

Epi

lanrat, 2022-03-25:
Thanks for the report.

The problem is due to an undocumented breaking change in the [Inkplate Library](https://github.com/e-radionicacom/Inkplate-Arduino-library) that I have yet to fully identify. For the time being I set HomePlate to use the last known good version while I investigate this. If you pull the latest code from this repo and try again it should fix this problem for now.

Unrelated to this problem, If you want to fix the waveform error see this new section in the [project readme](https://github.com/lanrat/homeplate#debugging).

closed 2022-03-25 by lanrat

C9H13NO3-dev, 2022-03-26:
Thanks you very much for solving this issue!