lanrat/homeplate #8: Scope Errors During PIO Run

wraith1385, 2022-07-17:
I get the following errors when I do PIO Run in VSCode:

Compiling .pio/build/esp32/src/time.cpp.o
src/image.cpp: In function 'bool hassImage()':
src/image.cpp:23:42: error: 'IMAGE_URL' was not declared in this scope
     uint8_t *buff = display.downloadFile(IMAGE_URL, &defaultLen);
                                          ^~~~~~~~~
src/info.cpp: In function 'void displayInfoScreen()':
src/info.cpp:212:20: error: 'MQTT_HOST' was not declared in this scope
     display.printf(MQTT_HOST);
                    ^~~~~~~~~
src/info.cpp:231:20: error: 'NTP_SERVER' was not declared in this scope
     display.printf(NTP_SERVER);
                    ^~~~~~~~~~
src/info.cpp:231:20: note: suggested alternative: 'SNTP_SERVER_DNS'
     display.printf(NTP_SERVER);
                    ^~~~~~~~~~
                    SNTP_SERVER_DNS
*** [.pio/build/esp32/src/image.cpp.o] Error 1
*** [.pio/build/esp32/src/info.cpp.o] Error 1
src/main.cpp: In function 'void setup()':
src/main.cpp:44:22: error: 'TOUCHPAD_ENABLE' was not declared in this scope
     if (sleepBoot && TOUCHPAD_ENABLE)
                      ^~~~~~~~~~~~~~~
src/main.cpp:44:22: note: suggested alternative: 'TOUCH_PAD_MAX'
     if (sleepBoot && TOUCHPAD_ENABLE)
                      ^~~~~~~~~~~~~~~
                      TOUCH_PAD_MAX
src/input.cpp: In function 'void checkButtons(void*)':
src/input.cpp:51:13: error: 'TOUCHPAD_ENABLE' was not declared in this scope
         if (TOUCHPAD_ENABLE)
             ^~~~~~~~~~~~~~~
src/input.cpp:51:13: note: suggested alternative: 'TOUCH_PAD_MAX'
         if (TOUCHPAD_ENABLE)
             ^~~~~~~~~~~~~~~
             TOUCH_PAD_MAX
*** [.pio/build/esp32/src/main.cpp.o] Error 1
*** [.pio/build/esp32/src/input.cpp.o] Error 1
src/network.cpp: In function 'void keepWiFiAlive(void*)':
src/network.cpp:63:26: error: 'HOSTNAME' was not declared in this scope
         WiFi.setHostname(HOSTNAME); // only works with DHCP....
                          ^~~~~~~~
src/network.cpp:63:26: note: suggested alternative: 'ESTALE'
         WiFi.setHostname(HOSTNAME); // only works with DHCP....
                          ^~~~~~~~
                          ESTALE
src/network.cpp:64:20: error: 'WIFI_SSID' was not declared in this scope
         WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
                    ^~~~~~~~~
src/network.cpp:64:20: note: suggested alternative: 'WIFI_STA'
         WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
                    ^~~~~~~~~
                    WIFI_STA
src/network.cpp:64:31: error: 'WIFI_PASSWORD' was not declared in this scope
         WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
                               ^~~~~~~~~~~~~
src/network.cpp:64:31: note: suggested alternative: 'WIFI_PS_NONE'
         WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
                               ^~~~~~~~~~~~~
                               WIFI_PS_NONE
src/ota.cpp: In function 'void startOTATask()':
src/ota.cpp:40:28: error: 'HOSTNAME' was not declared in this scope
     ArduinoOTA.setHostname(HOSTNAME);
                            ^~~~~~~~
src/ota.cpp:40:28: note: suggested alternative: 'ESTALE'
     ArduinoOTA.setHostname(HOSTNAME);
                            ^~~~~~~~
                            ESTALE
src/qr.cpp: In function 'void displayWiFiQR()':
src/qr.cpp:11:51: error: 'QR_WIFI_NAME' was not declared in this scope
     snprintf(buf, 1024, "WIFI:S:%s;T:WPA;P:%s;;", QR_WIFI_NAME, QR_WIFI_PASSWORD);
                                                   ^~~~~~~~~~~~
src/qr.cpp:11:51: note: suggested alternative: 'WIFI_AP'
     snprintf(buf, 1024, "WIFI:S:%s;T:WPA;P:%s;;", QR_WIFI_NAME, QR_WIFI_PASSWORD);
                                                   ^~~~~~~~~~~~
                                                   WIFI_AP
src/qr.cpp:11:65: error: 'QR_WIFI_PASSWORD' was not declared in this scope
     snprintf(buf, 1024, "WIFI:S:%s;T:WPA;P:%s;;", QR_WIFI_NAME, QR_WIFI_PASSWORD);
                                                                 ^~~~~~~~~~~~~~~~
*** [.pio/build/esp32/src/network.cpp.o] Error 1
src/sleep.cpp: In function 'void gotoSleepNow()':
src/sleep.cpp:31:9: error: 'TOUCHPAD_ENABLE' was not declared in this scope
     if (TOUCHPAD_ENABLE)
         ^~~~~~~~~~~~~~~
src/qr.cpp:11:65: note: suggested alternative: 'WIFI_PS_NONE'
     snprintf(buf, 1024, "WIFI:S:%s;T:WPA;P:%s;;", QR_WIFI_NAME, QR_WIFI_PASSWORD);
                                                                 ^~~~~~~~~~~~~~~~
                                                                 WIFI_PS_NONE
src/sleep.cpp:31:9: note: suggested alternative: 'TOUCH_PAD_MAX'
     if (TOUCHPAD_ENABLE)
         ^~~~~~~~~~~~~~~
         TOUCH_PAD_MAX
src/sleep.cpp:40:9: error: 'TOUCHPAD_ENABLE' was not declared in this scope
     if (TOUCHPAD_ENABLE)
         ^~~~~~~~~~~~~~~
src/sleep.cpp:40:9: note: suggested alternative: 'TOUCH_PAD_MAX'
     if (TOUCHPAD_ENABLE)
         ^~~~~~~~~~~~~~~
         TOUCH_PAD_MAX
*** [.pio/build/esp32/src/ota.cpp.o] Error 1
src/time.cpp: In function 'void ntpSync(void*)':
src/time.cpp:23:34: error: 'NTP_SERVER' was not declared in this scope
     NTPClient timeClient(ntpUDP, NTP_SERVER);
                                  ^~~~~~~~~~
src/time.cpp:23:34: note: suggested alternative: 'SNTP_SERVER_DNS'
     NTPClient timeClient(ntpUDP, NTP_SERVER);
                                  ^~~~~~~~~~
                                  SNTP_SERVER_DNS
*** [.pio/build/esp32/src/qr.cpp.o] Error 1
*** [.pio/build/esp32/src/sleep.cpp.o] Error 1
*** [.pio/build/esp32/src/time.cpp.o] Error 1
src/mqtt.cpp: In function 'void startMQTTTask()':
src/mqtt.cpp:442:26: error: 'HOSTNAME' was not declared in this scope
   mqttClient.setClientId(HOSTNAME);
                          ^~~~~~~~
src/mqtt.cpp:442:26: note: suggested alternative: 'ESTALE'
   mqttClient.setClientId(HOSTNAME);
                          ^~~~~~~~
                          ESTALE
src/mqtt.cpp:443:24: error: 'MQTT_HOST' was not declared in this scope
   mqttClient.setServer(MQTT_HOST, MQTT_PORT);
                        ^~~~~~~~~
src/mqtt.cpp:443:35: error: 'MQTT_PORT' was not declared in this scope
   mqttClient.setServer(MQTT_HOST, MQTT_PORT);
                                   ^~~~~~~~~
src/mqtt.cpp:443:35: note: suggested alternative: 'NOT_A_PORT'
   mqttClient.setServer(MQTT_HOST, MQTT_PORT);
                                   ^~~~~~~~~
                                   NOT_A_PORT
*** [.pio/build/esp32/src/mqtt.cpp.o] Error 1

lanrat, 2022-07-17:
It looks like your config is not set correctly. Copy `src/config_example.h` to `src/config.h` and edit the settings and then run `pio run` again.

wraith1385, 2022-07-17:
Thanks, I stupidly had config.h up one level instead of in the src folder.

closed 2022-07-17 by wraith1385