lanrat/homeplate #14: Activity deadlock when action is "hass"

elratt0r, 2022-12-05:
When having a mqtt message pending with `{"action": "hass"}` and the homeplate boots from timer/wake button, it will start up with activity HomeAssistant, connect to the WiFi, pass the [waitForWiFiOrActivityChange()](https://github.com/lanrat/homeplate/blob/main/src/activity.cpp#L90) but then during run of remotePNG gets aborted by the [stopActivity()](https://github.com/lanrat/homeplate/blob/main/src/image.cpp#L37) because [resetActivity = true](https://github.com/lanrat/homeplate/blob/main/src/activity.cpp#L27) is set by the incoming mqtt message. The pending activity from mqtt is not run because [debounce conditions](https://github.com/lanrat/homeplate/blob/main/src/activity.cpp#L73) apply.

I think it should get caugth [here](https://github.com/lanrat/homeplate/blob/main/src/activity.cpp#L19) instead, but because activityCurrent gets [set](https://github.com/lanrat/homeplate/blob/main/src/activity.cpp#L132) after the activity is done running, it never applies.

Two ideas for solutions:
Set activityCurrent = activityNext right after accepting the activity to be run ([here](https://github.com/lanrat/homeplate/blob/main/src/activity.cpp#L78))

Checking resetActivity after running remotePNG to check if it was aborted and then use "continue" in the while loop instead of falling through and setting activityCurrent = activityNext.

elratt0r, 2022-12-05:
Patch for solution 1 [here](/elratt0r/homeplate/tree/fix_reset_activity)

lanrat, 2022-12-10:
Good find.

I'm fine with either of your proposed solutions. If you send a PR for either I'll merge it.

commit 5eea8526 (2022-12-10): Merge pull request #15 from elratt0r/fix_reset_activity

set activityCurrent earlier to avoid resetActivity deadlock

Fixes #14.
closed 2022-12-10 by lanrat