lanrat/homeplate #26: Timezone always stuck in one timezone

elratt0r, 2023-04-02:
Because epoch is initialized with 0 here
 https://github.com/lanrat/homeplate/blob/e6e3cf25aef40a748c899e4332ddc93a367a5f13/src/time.cpp#L14
the tzOffset always returns the first DST-Parameter.
tzOffset should be called with the current unix time, and then returning the matching offset to put in rtc.offset:

```
time_t tzOffset(time_t epoch) {
    TimeChangeRule *tcr;
    time_t offset = epoch - tz.toLocal(epoch, &tcr);
    return offset
}

rtc.offset = tzOffset(rtc.getLocalEpoch());
```

Furthermore, the i2c rtc is not used anymore to set the esp32 time.
I would suggest to set ESP32 RTC from i2c RTC on startup.

```
epoch = display.rtcGetEpoch();
rtc.setTime(epoch)
```

lanrat, 2023-04-05:
I fixed the timezone issue, but the internal hardware RTC still does not increment on its own.
As long as every fresh boot can perform NTP its not an issue.

closed 2023-04-05 by lanrat

elratt0r, 2023-04-13:
thx, lgtm
Currently running a test version to monitor the deviation of external rtc, internal rtc and ntp over time. Will keep you posted.

lanrat, 2023-04-13:
@elratt0r does your HW RTC increment? Mine no longer does.

For example, if I set my RTC with an epoch of `123456,` sleep for 5 seconds, and then query the RTC for the latest time, I get `123456` again. This did not used to be the case.

I'm wondering if my RTC failed (HW issue) or if something in the updated Inkplate library broke it?

commit c070f97b (2023-04-05): fixes #26, timezone not being set