lanrat/homeplate #27: read only dashboard configuration

NullReferenceError, 2023-04-08:
Hey, this inkplate program is great! I've been tinkering around with it, and have everything loading properly, but haven't found a good way to bring a read only Dashboard from Home Assistant in - is there a suggested method for this? I have a basic dashboard created using lovelace card mod, like some of the examples here, but can't find a good resource on making only this dashboard exposed. Or, is there a method with the current API call to include credentials or API token when we try to hit home assistant?

lanrat, 2023-04-09:
Does home assistant support read-only dashboards?

All Homeplate does is display a static PNG, which is read-only be definition. It works with a program like [hass-screenshot](https://github.com/lanrat/hass-screenshot) which will need a token to gain access to the home assistant instance.

NullReferenceError, 2023-04-14:
ah - thanks! I was able to get that setup and working now. I noticed some of the mqtt entities are not showing up in home assistant, maybe I should open a different issue for this, but is this something you have noticed as well?

closed 2023-04-14 by NullReferenceError