lanrat/homeplate #30: Compile error NTP_SYNC_INTERVAL

airmaxx, 2023-05-29:
I tried to update my installation with

```
git pull
pio upgrade
pio pkg update
pio run --target clean
pio run
```

I get the following compile error
```
Compiling .pio/build/ota/lib789/WiFi/WiFiClient.cpp.o
src/time.cpp: In function 'void setupTimeAndSyncTask()':
src/time.cpp:102:38: error: 'NTP_SYNC_INTERVAL' was not declared in this scope
         bool resync = ((bootCount % (NTP_SYNC_INTERVAL)) == 0);
                                      ^~~~~~~~~~~~~~~~~
src/time.cpp:102:38: note: suggested alternative: 'TCP_TMR_INTERVAL'
         bool resync = ((bootCount % (NTP_SYNC_INTERVAL)) == 0);
                                      ^~~~~~~~~~~~~~~~~
                                      TCP_TMR_INTERVAL
*** [.pio/build/ota/src/time.cpp.o] Error 1
```

lanrat, 2023-05-29:
`NTP_SYNC_INTERVAL` was a recently added config parameter.

Please see the defines in `config_example.h` and copy them to your `config.h` and try compiling again.

lanrat, 2023-07-03:
Closing due to inactivity. If you are still having issues, let me know.

closed 2023-07-03 by lanrat