lanrat/homeplate #34: Sleep of 440min gets disrupted

tuxianerDE, 2023-11-18:
Good morning,

I made the following observation, for the night time I have HA send an mqtt message to the inkplate-10 with a message action, content and a refresh of 440min. However I noticed that the inkplate only sleeps for about 10min and wakes up again.

I the device console it properly logged when I tested it the 440min as interval and also 10min is not configured anywhere (default is 20min). So I was wondering, since I am below the limit of 1 day as I looked through the code it should allow for that long period.

What am I missing?

Cheers
Tuxianer

lanrat, 2023-11-25:
Have you checked the serial logs when the device wakes from sleep? The boot reason should be one of the first few lines printed.

tuxianerDE, 2023-11-25:
Could not check that so far (on the todo list) however the activity posted on MQTT states timer. 

I did check on console when setting the sleep limit that at the end the 440min got properly printed (and thus also properly converted to minutes from the seconds I entered)

lanrat, 2023-11-25:
Do you have the Inkplate 10 v1 or v2? and do you have a RTC battery setup?

Sometimes, this can cause incorrect sleep times.

tuxianerDE, 2023-11-26:
v2 is used here and I have an RTC battery setup. I have inserted the RTC battery (this little cell) at the back and pushed until it made a click. Is there a way to verify if it working properly or at least correctly equipped?

PR #36 (2023-11-27): add datatype to factor to get uint64_t
See esp_sleep_enable_timer_wakeup(uint64_t time_in_us)

With uint32_t a maximum sleep time of ~4294s/71min is possible, before the counter wraps around.

Fixes #34
closed 2023-11-27 by lanrat