lanrat/homeplate #39: Sleep

abbottcam, 2024-04-13:
Hi there

quick question...// How long to sleep between image refreshes
#define TIME_TO_SLEEP_MIN 20. So am I right in thinking...It refreshes, then sleeps for 20 minutes? or refreshes and sleeps after 20 minutes?

Thanks

lanrat, 2024-04-13:
Your assumption is correct.

You can disable sleep mode, but don't expect your battery to last more than a few days, likely less. If you plan to run it with no battery and always on external power then it could work.

There is no way for Home Assistant (or any external service) to wake the device while it is in sleep mode as it is not connected to the network at that time.

Alternatively, you can send the `refresh` MQTT variable from home assistant to dynamically set the NEXT sleep cycle duration.  More information here: https://github.com/lanrat/homeplate/blob/main/hass.md

closed 2024-04-13 by lanrat

abbottcam, 2024-04-14:
ok thanks!

Ive set the sleep for 30 minutes, but sometimes it fails to refresh...Im not sure it's coming out of deep sleep. the internal clock doesnt seem to be hold time 

17:21:33.672 > [TIME] Internal Clock and RTC differ by -1713111693 seconds. local(0) RTC(1713111693)
17:21:33.681 > [TIME] local time (0) 01:00:00 Thursday, January 01 1970
17:21:33.686 > [TIME] ERROR: RTC time is too far in past. RTC likely has wrong value!

lanrat, 2024-04-14:
The RTC having the correct time is not a requirement for sleep mode to work.

RTC gets set from NTP if you set that up. But it is not required.