lanrat/homeplate #40: inkplate 10 doesn't show any image

huggenknubbel, 2024-04-29:
Hello,
got mr Inkplate 10 today, seting all up but it doesn't show any picture. 
still have the "welcome to ihnkplate" screen.

The Logs said: 
```
21:23:56.558 > [IMAGE] Downloading image: http://192.168.2.28:5000/1.png
21:23:56.563 > [  9036][E][WiFiClient.cpp:320] setSocketOption(): fail on -1, errno: 9, "Bad file number"
21:23:56.569 > [MQTT] Sending MQTT State: [homeassistant/sensor/homeplate/battery/state] {"voltage":0.28,"battery":0}
21:23:56.766 > [IMAGE] Download done
21:23:56.766 > [STATUS] Rendering image...
21:23:57.801 > [ 10282][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:23:58.827 > [ 11307][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:24:02.692 > [IMAGE] Image render ready
21:24:02.692 > [IMAGE] displaying....
21:24:03.703 > [ 16184][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:24:03.954 > [IMAGE] displaying done.
```
The MQTT values have arrived in home assistant. Beside the temperature. 
(maye thats the i2cRead error?)

elratt0r, 2024-04-30:
Are you compiling with inkplate10v2? The i2cRead errors point to a problem talking to the IO Expander or the PSU for the ePaper.

huggenknubbel, 2024-04-30:
Tried it with v2 too. but there i didn't get any connection to the inkplate after flashing:
```
Wrote 1292496 bytes (759139 compressed) at 0x00010000 in 18.9 seconds (effective 548.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Disconnected (ClearCommError failed (PermissionError(13, 'Zugriff verweigert', None, 5)))
Reconnecting to COM4
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html
```

got the "permission denied" entry every couple seconds until i cancel with strg+c

maybe its an issue beceause of an other chip they use since vers.1.3? 
i found this [issue ](https://github.com/esphome/issues/issues/4287) with esphome.

elratt0r, 2024-04-30:
The PCAL6414 on the Inkplate10v2 is in the library that is used by homeplate
https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/7.0.0/src/include/Pcal.cpp

Could you post your config.h?
And for the COM4/Zugriff verweigert: Is the Inkplate visible after flashing in the device manager on the same COM port?

huggenknubbel, 2024-04-30:
the Serial Port is another wierd thing if i'm flashing with V2 settings (doesn't have this behavior with "normal" flash) i attached a little GIF to visualize this.
My config.h looks like:

```
#pragma once

// WiFi SSID
#define WIFI_SSID "Schneckenhaus" 
// WiFi password
#define WIFI_PASSWORD "mypassword"

// hostname
// NOTE: if using multiple homeplate devices, you MUST make the hostname unique
#define HOSTNAME "homeplate"

// How long to sleep between image refreshes
#define TIME_TO_SLEEP_MIN 1

// Static IP information
// If unset uses DHCP, but updates may be slower, set to use a Static IP
// #define STATIC_IP "192.168.1.10"
// #define STATIC_SUBNET "255.255.255.0"
// #define STATIC_GATEWAY "192.168.1.1"
// #define STATIC_DNS "192.168.1.1"

// NTP Time server to set RTC
#define NTP_SERVER "130.149.17.21"

// How often to re-sync the clock to NTP
#define NTP_SYNC_INTERVAL (24*60)/TIME_TO_SLEEP_MIN // ~ once a day when updating every TIME_TO_SLEEP_MIN minutes

// URL of PNG image to display
#define IMAGE_URL "http://192.168.2.28:5000/1.png"

// WiFi QR Code
#define QR_WIFI_NAME "Schneckenhaus"
#define QR_WIFI_PASSWORD "Guest WiFi Password"

// MQTT Broker
#define MQTT_HOST "192.168.2.74"
#define MQTT_PORT 1883
// Set MQTT_USER & MQTT_PASSWORD if needed
#define MQTT_USER "Huggenknubbel"
#define MQTT_PASSWORD "mypassword"
// Customize node id and device name if needed
//#define MQTT_NODE_ID "homeplate"	// defaults to HOSTNAME
//#define MQTT_DEVICE_NAME "HomePlate"	// defaults to "HomePlate"

// Disables touchpads if they are overly sensitive and result in phantom touch events
// Touchpads are not supported on the Inkplate10v2. This must be false if ARDUINO_INKPLATE10V2 is set
#define TOUCHPAD_ENABLE false

// Timezone
// see timezone_config.h for options
#define TIMEZONE_UTC

// If your Inkplate doesn't have external (or second) MCP I/O expander, you should uncomment next line,
// otherwise your code could hang out when you send code to your Inkplate.
// You can easily check if your Inkplate has second MCP by turning it over and 
// if there is missing chip near place where "MCP23017-2" is written, but if there is
// chip soldered, you don't have to uncomment line and use external MCP I/O expander
#define ONE_MCP_MODE

// Displays the time from the RTC whenever a new image is loaded
#define DISPLAY_LAST_UPDATE_TIME false

// keep this to signal the program has a valid config file
#define CONFIG_H
```

https://github.com/lanrat/homeplate/assets/15655280/fdffc0bc-e9f4-4f9f-9bfe-0c7c1f5d3301

elratt0r, 2024-04-30:
Could you try to build debug, once with -DARDUINO_INKPLATE10 and once with -DARDUINO_INKPLATE10V2
(https://github.com/lanrat/homeplate/blob/4d50ca996f47e8e20787bbf60eeef125b4ea2d14/platformio.ini#L64C3-L64C23)

huggenknubbel, 2024-04-30:
Debug Output -DARDUINO_INKPLATE10

```
Wrote 1361056 bytes (798098 compressed) at 0x00010000 in 19.8 seconds (effective 551.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
13:15:07.590 > ets Jul 29 2019 12:21:46
13:15:07.590 >
13:15:07.590 > rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
13:15:07.590 > configsip: 0, SPIWP:0xee
13:15:07.590 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:15:07.590 > mode:DIO, clock div:2
13:15:07.590 > load:0x3fff0030,len:1184
13:15:07.590 > load:0x40078000,len:13232
13:15:07.590 > load:0x40080400,len:3028
13:15:07.590 > entry 0x400805e4
13:15:07.590 > [     8][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
13:15:07.590 > [   457][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
13:15:07.590 >
13:15:07.590 >
13:15:07.590 > [SETUP] starting, version(Apr 30 2024, 13:14:21) boot: 0
13:15:07.590 > ESP Chip information:
13:15:07.590 > This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 3, 4MB external flash
13:15:07.590 > Minimum free heap size: 4414999 bytes
13:15:07.590 > Heap summary for capabilities 0x00000C06:
13:15:07.590 >   Totals:
13:15:07.590 >     free 0 allocated 0 min_free 0 largest_free_block 0
13:15:07.590 > [ACTIVITY] startActivity(1) put into queue
13:15:07.590 > [   505][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
13:15:07.590 > [STATUS] Boot 1 normal
13:15:07.909 > [  1642][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
13:15:08.935 > [  2668][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
13:15:08.944 > [  2676][D][esp32-hal-adc.c:194] __analogReadMilliVolts(): eFuse Vref: Supported
13:15:08.949 > [  2676][I][esp32-hal-adc.c:235] __analogReadMilliVolts(): ADC1: Characterized using eFuse Vref: 1093
13:15:08.957 >
13:15:08.957 > [SETUP] Battery: 0% (0.28v)
13:15:10.005 > [  3738][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
13:15:10.011 > [SETUP] starting button task
13:15:10.014 > [SETUP] starting time task
13:15:10.014 > [TIME] local time (3) 00:00:03 Thursday, January 01 1970
13:15:10.019 > [SETUP] starting WiFi task
13:15:10.022 > [WIFI] Connecting...
13:15:10.025 > [SETUP] starting OTA task
13:15:10.028 > [SETUP] starting MQTT task
13:15:10.031 > [MQTT] starting MQTT
13:15:10.031 > [SETUP] starting sleep task
13:15:10.034 > [SETUP] starting activities task
13:15:10.037 > [ACTIVITY] starting activity: 1
13:15:10.040 > [  3767][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
13:15:10.103 > [  3834][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
13:15:10.170 > [  3902][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
13:15:10.175 > [WIFI] Connected to AP successfully!
13:15:10.200 > [  3932][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
13:15:10.205 > [  3932][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 192.168.2.109, MASK: 255.255.255.0, GW: 192.168.2.1
13:15:10.217 > [WIFI] IP address: 192.168.2.109
13:15:10.220 > [STATUS] WiFi connected
13:15:10.235 > [WIFI] Connected: 192.168.2.109
13:15:10.269 > [TIME] Syncing RTC to NTP
13:15:10.282 > [MQTT] Connecting to MQTT...
13:15:10.282 > [  4014][I][AsyncMqttClient.cpp:686] connect(): CONNECTING
13:15:10.287 > [SLEEP] delaying sleep for 20 seconds
13:15:10.293 > [STATUS] Downloading image...
13:15:10.293 > [  4020][I][ArduinoOTA.cpp:141] begin(): OTA server at: homeplate.local:3232
13:15:10.301 > [OTA] OTA ready
[... 178 lines cut ...]
13:15:32.111 > [ 25838][I][AsyncMqttClient.cpp:224] _onDisconnect(): TCP disconn
13:15:32.119 > 
13:15:32.119 > [MQTT] Disconnected from MQTT.
13:15:32.122 > [ 25854][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
13:15:32.128 > [ 25855][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 8 - ASSOC_LEAVE
13:15:32.137 > [WIFI] Disconnected from WiFi access point
13:15:32.139 > [WIFI] WiFi lost connection. Reason: 8
13:15:32.142 > [ 25869][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 3 - STA_STOP
13:15:32.155 > [SLEEP] entering sleep for 1200 seconds (20 min)
```

Debug Output with -DARDUINO_INKPLATE10V2 doesn't change very much:
```
Wrote 1360432 bytes (798163 compressed) at 0x00010000 in 19.8 seconds (effective 548.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Disconnected (ClearCommError failed (PermissionError(13, 'Zugriff verweigert', None, 5)))
Reconnecting to COM4     Connected!
```

elratt0r, 2024-04-30:
Could you try a demo project from the Inkplate Library? As the thermistor is in the PMIC (TPS65186) of the eInk and this is written via I2C during the bootup I suspect there's something wrong talking to that one in particular.
I still have no clue why V2 behaves that different :-(

huggenknubbel, 2024-05-01:
i tried some examples and they didn't run either.
write an email to the support. Maybe my device is broken?

elratt0r, 2024-05-01:
Yes, sounds like sth. is wrong with the device unfortunately.

huggenknubbel, 2024-05-03:
I'm found the issue. and i'm speachless about it. 
Today the Battery which i ordered arrived and i connected  it. What i should say? It works as expected. 
Wow.
Thanks @elratt0r for you help anyway!

closed 2024-05-07 by huggenknubbel