lanrat/homeplate #42: Inkplate 10 'Bad file number' error on image download

tommyalatalo, 2024-05-09:
I'm trying to set up an Inkplate 10 (v2), and I'm hosting a home assistant dashboard screenshot using `sibbl/hass-lovelace-kindle-screensaver` with the below configuration.

The firmware is successfully flashed to the device and the display shows "HomePlate" after flashing, but when the process tries to download the image from the server I get the error below. Why is this? Is the image server misconfigured somehow? I'm not sure what to make of the message `Bad file number`. I can successfully download the png image from the server using wget, curl or a web browser.

Full log from startup: https://0x0.st/X82m.txt

```
09:16:54.937 > [IMAGE] Downloading image: http://hass-inkplate.example.com
09:16:54.940 > [  6942][E][WiFiClient.cpp:320] setSocketOption(): fail on -1, errno: 9, "Bad file number"
09:16:59.960 > [IMAGE] Download failed
09:16:59.960 > [STATUS] Download failed!
```

```
HA_BASE_URL=https://homeassistant.example.com
HA_SCREENSHOT_URL=/dashboard-inkplate/0?kiosk
HA_ACCESS_TOKEN={{}}
CRON_JOB=* * * * *
RENDERING_TIMEOUT=30000
RENDERING_DELAY=0
RENDERING_SCREEN_HEIGHT=820
RENDERING_SCREEN_WIDTH=1200
GRAYSCALE_DEPTH=5
OUTPUT_PATH=/output
LANGUAGE=en
ROTATION=0
SCALING=1
```

lanrat, 2024-05-09:
I think that your `IMAGE_URL` in `config.h` is incorrect. It should be a full URL to the png file, not just the domain of the screenshot tool.

`config.h` example assuming the path is `/kiosk.png`:

```c
#define IMAGE_URL "http://hass-inkplate.example.com/kiosk.png"
```

tommyalatalo, 2024-05-10:
> I think that your `IMAGE_URL` in `config.h` is incorrect. It should be a full URL to the png file, not just the domain of the screenshot tool.
> 
> `config.h` example assuming the path is `/kiosk.png`:
> 
> ```c
> #define IMAGE_URL "http://hass-inkplate.example.com/kiosk.png"tv

> ```

Yeah it doesn't seem like `sibbl/hass-lovelace-kindle-screensaver` exposes a png file explicitly like that.
When I inspect the source on the page that the screensaver container exposes the image is defined like this:
```
<img src="https://hass-inkplate.example.com/ alt="https://hass-inkplate.example.com/" class="">
```
So there is no `.png` target to go to?

lanrat, 2024-05-10:
I believe you are correct. If you want to verify that the issue is with the png url you can try using another png url from another website.

I use my fork of `sibbl/hass-lovelace-kindle-screensaver` https://github.com/lanrat/hass-screenshot, which acts a little differently, and since the fork the projects have diverged some.

I am sure it is still possible to get it working with sibbl's project, but I likely can't help you debug that as I don't use it.
If you do get it working, please report back so that others can benefit. Or you can try using my fork.

tommyalatalo, 2024-05-10:
> I believe you are correct. If you want to verify that the issue is with the png url you can try using another png url from another website.
>
> I use my fork of `sibbl/hass-lovelace-kindle-screensaver` [...]

I switched over to your fork, but I'm still getting the same error, logs at https://0x0.st/X80o.txt

I'm running the your fork of the screenshot service with this config:
```
HA_BASE_URL=https://homeassistant.example.com
HA_SCREENSHOT_URL=/dashboard-inkplate/0?kiosk
HA_ACCESS_TOKEN={{ .Data.data.screensaver_access_token }}
# CRON_JOB=* * * * *
RENDERING_TIMEOUT=30000
RENDERING_DELAY=2
RENDERING_SCREEN_HEIGHT=820
RENDERING_SCREEN_WIDTH=1200
COLOR_MODE=GrayScale
GRAYSCALE_DEPTH=5
# OUTPUT_PATH=/output
LANGUAGE=en
REAL_TIME=true
# ROTATION=0
# SCALING=1
```

lanrat, 2024-05-10:
If you view `https://hass-inkplate.example.com/1.png` in your browser do you get a valid png file?

The error `[E][WiFiClient.cpp:320] setSocketOption(): fail on -1, errno: 9, "Bad file number"` is safe to ignore. That's being thrown by the underlying library all the time, even when it works.

tommyalatalo, 2024-05-11:
Yes, I do get the image when going to that URL, it opens up as expected in the browser and I can download it manually from there.

tommyalatalo, 2024-05-11:
I also tried changing the image URL to IP:Port instead like below, but it still just fails with "Download failed" for some reason.
Downloading the image using the IP:Port address works just fine with wget etc.

```
08:54:09.512 > [IMAGE] Downloading image: http://192.168.10.10:27277/1.png
08:54:09.515 > [  6946][E][WiFiClient.cpp:320] setSocketOption(): fail on -1, errno: 9, "Bad file number"
08:54:09.523 > [MQTT] Sending MQTT State: [homeassistant/sensor/homeplate/battery/state] {"voltage":4.26,"battery":100}
08:54:14.519 > [IMAGE] Download failed
08:54:14.519 > [STATUS] Download failed!
```

mmmmmtasty, 2024-06-22:
I see this exact same issue, also with an Inkplate 10, although I have a couple more lines of output. I'm using the sbbl screenshotter, and I have a working inkplate6color instance using the same setup so I know it works. I get the following:

`22:54:40.508 > [SETUP] starting, version(Jun 21 2024, 22:47:34) boot: 1
22:54:40.511 > [ACTIVITY] startActivity(1) put into queue
22:54:40.516 > Wavefrom load failed! Upload new waveform in EEPROM. Using default waveform.
22:54:40.642 > [SETUP] Battery: 0% (0.28v)
22:54:40.642 > [SETUP] starting time task
22:54:40.645 > [TIME] Internal Clock and RTC differ by 1 seconds. local(1719024881) RTC(1719024880)
22:54:40.653 > [TIME] local time (1719024881) 22:54:41 Friday, June 21 2024
22:54:40.659 > [SETUP] starting WiFi task
22:54:40.661 > [WIFI] Connecting...
22:54:40.661 > [SETUP] starting OTA task
22:54:40.664 > [SETUP] starting MQTT task
22:54:40.667 > [MQTT] starting MQTT
22:54:40.670 > [SETUP] starting sleep task
22:54:40.672 > [SETUP] starting activities task
22:54:40.675 > [ACTIVITY] starting activity: 1
22:54:40.678 > [SLEEP] delaying sleep for 15 seconds
22:54:40.838 > [WIFI] Connected to AP successfully!
22:54:41.023 > [WIFI] IP address: 192.168.1.227
22:54:41.098 > [WIFI] Connected: 192.168.1.227
22:54:41.169 > [MQTT] Connecting to MQTT...
22:54:41.171 > [OTA] OTA ready
22:54:41.180 > [SLEEP] delaying sleep for 240 seconds
22:54:41.183 > [IMAGE] Downloading image: http://homeassistant.local:5006/2.png
22:54:41.298 > [  1271][E][WiFiClient.cpp:320] setSocketOption(): fail on -1, errno: 9, "Bad file number"
22:54:41.336 > [MQTT] Connected to MQTT.
22:54:41.337 > [MQTT] Session present: 0
22:54:41.339 > [MQTT] Subscribing at QoS 2, packetId: 1
22:54:41.344 > [MQTT] Sending MQTT Config
22:54:41.359 > [MQTT] Subscribe acknowledged: packetId: 1 qos: 2 
22:54:41.370 > [MQTT] Connected
22:54:42.669 > [MQTT] sending status update
22:54:42.670 > [MQTT] Sending MQTT State: [homeassistant/sensor/inkplate10/boot/state] {"boot":2,"activity_count":1,"boot_reason":"wake button"}
22:54:42.882 > [MQTT] Sending MQTT State: [homeassistant/sensor/inkplate10/wifi_signal/state] {"signal":-71}
22:54:43.899 > [  3871][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
22:54:43.905 > [MQTT] Got invalid temperature (-1), not sending status to mqtt
22:54:43.913 > [MQTT] Sending MQTT State: [homeassistant/sensor/inkplate10/battery/state] {"voltage":0.28,"battery":0}
22:55:01.872 > [IMAGE] displaying....
22:55:02.891 > [ 22863][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
22:55:03.141 > [IMAGE] displaying done.`

hillbillypolenta, 2024-07-12:
> I see this exact same issue, also with an Inkplate 10, although I have a couple more lines of output. I'm using the sbbl screenshotter, and I have a working inkplate6color instance using the same setup so I know it works. I get the following:

Go into config.h and uncomment out the "#define ONE_MCP_MODE" and build with "pio run -e inkplate10v2"

I had exactly the same error 263 and whatnot.

felixh10r, 2024-07-13:
I've encountered this problem after switching from [sibbl/hass-lovelace-kindle-screensaver](https://github.com/sibbl/hass-lovelace-kindle-screensaver/issues) to [lanrat/hass-screenshot](https://github.com/lanrat/hass-screenshot) (because of the CPU-saving `REAL_TIME` option).

I can report that in my case, the `REAL_TIME` option was the problem, maybe because the image rendering took longer than the request timeout? When loading a cached image (as per the `REAL_TIME_CACHE_SEC` option), the problem didn't occur.

I'm not experienced enough to provide a PR, but I'm happy to add more details in case I can help.

mmmmmtasty, 2024-07-13:
> Go into config.h and uncomment out the "#define ONE_MCP_MODE" and build with "pio run -e inkplate10v2"

This worked for me, thanks!

hillbillypolenta, 2024-07-14:
> I've encountered this problem after switching from [sibbl/hass-lovelace-kindle-screensaver](https://github.com/sibbl/hass-lovelace-kindle-screensaver/issues) to [...]

If the PNG isn't available or not displayable at the time the draw process runs you should be seeing "Image display error" or "Download failed" in the serial output. I'm using REAL_TIME and haven't had any issues, but the docker instance I'm running hass-screenshot on is on an SSD which greatly improves performance.

Possible feature request for the routine to pause and retry n times on download or render fail?

felixh10r, 2024-07-14:
I also get a "Download failed" message, but I thought it was closely related to the "Bad file number" message.
The screenshot instance runs on an SSD and renders quite instantly, so I'm not quite sure what causes the failed download.

A feature for inkplate to pause and retry sounds good, however it would even solve my particular problem when the image load could be timed more exactly; e.g. my goal is to display new data every 5 minutes, so in theory I could just create a new screenshot every 5 minutes and load the image soon after. However, while the inkplate supports a sleep time (e.g. 5000), due to the seconds the startup procedure takes, the time the image is actually loaded is drifting 🤔

hillbillypolenta, 2024-07-14:
> "Bad file number" message.

I would try the Inkplate10v2 env as described above in my reply to mmmmmtasty

tommyalatalo, 2024-07-14:
> > "Bad file number" message.
> 
> I would try the Inkplate10v2 env as described above in my reply to mmmmmtasty

I followed your suggestion, and also tinkered with the `REAL_TIME` setting, none of it fixes my inital problem unfortunately.

felixh10r, 2024-07-14:
Thanks hillbillypolenta—sadly, I'm already using the Inkplate10v2 env and the `ONE_MCP_MODE` config didn't solve the problem either 🤔

hillbillypolenta, 2024-07-14:
Sorry I can't be of more help, that's kind of the extent of my knowledge. I'll include my lanrat/hass-screenshot config here for reference in case it is helpful. Checking in a browser, the refresh process of hass-screenshot takes about 3 seconds. I could probably drop the delay some, but it works fine currently.

`HA_BASE_URL http://192.168.1.100:8123
HA_SCREENSHOT_URL /lovelace-inkplate
HA_ACCESS_TOKEN <token>
PORT 5123
REAL_TIME true
COLOR_MODE GrayScale
GRAYSCALE_DEPTH 4
SCALING 1.55
RENDERING_SCREEN_WIDTH 1200
RENDERING_SCREEN_HEIGHT 820
RENDERING_DELAY 1500
RENDERING_TIMEOUT 5000
TZ <tz>
REAL_TIME_CACHE_SEC 60
REMOVE_GAMMA true
ROTATION 270`

mmmmmtasty, 2024-07-14:
FWIW, I still get the bad file number errors in the log, but the download
and display of the image succeeds which is all I really care about.

lanrat, 2024-08-01:
For what it's worth, I've always had the following in my logs, and its still always worked:

```
21:12:42.524 > [  4501][E][WiFiClient.cpp:313] setSocketOption(): fail on -1, errno: 9, "Bad file number"
```

I suspect if your image is not displaying, it is likely unrelated to this console error.

elratt0r, 2024-10-21:
> For what it's worth, I've always had the following in my logs, and its still always worked:
> 
> ```
> 21:12:42.524 > [  4501][E][WiFiClient.cpp:313] setSocketOption(): fail on -1, errno: 9, "Bad file number"
> ```
> 
> I suspect if your image is not displaying, it is likely unrelated to this console error.

Followed down that error, comes from setting TCP_NODELAY in the HTTPClient [Inkplate-Arduino-library/src/include/NetworkClient.cpp](https://github.com/SolderedElectronics/Inkplate-Arduino-library/blob/d33b0e7797eb42fdec34faf164216b547d32cbe3/src/include/NetworkClient.cpp#L413)

Doesn't do any harm so far. Will report upstream.

lanrat, 2025-02-28:
Closing this since its in the underlying library and is not the cause of an issue.

closed 2025-02-28 by lanrat