lanrat/homeplate #48: Inserted QR_WIFI_PASSWORD is too long to show

Gloomyeye, 2024-07-24:
Hello,

thank you for your great project and your effort.
If I try to see the QR_WIFI_PASSWORD (17 characters long) I have entered, I have to realize that it is displayed over the edge of the display and extends into the QR code. The password is not readble as well as the qr code is can not be used correctly. Unfortunately, I don't see any way to change the font size of the password so that the password can be displayed correctly. It is possible to add this feature inside the config.h? Alternatively, it is possible to add a logic which counts the characters and decrease the font size automatically.

Thank you!

lanrat, 2024-08-01:
There is existing code in https://github.com/lanrat/homeplate/blob/main/src/message.cpp#L27 that dynamically adjusts the font size so that a message can fit within the display area. It currently is only used by the message activity, and only for text that takes up the entire screen.

I don't think it would be too much work to modify it to take a variable screen area as an additional input and use it for the wifi activity as well.

I would welcome a PR with this enhancement if you or anyone else wants to tackle it.

Gloomyeye, 2024-08-01:
Thank you for your hints. However, I am not familiar with C or C++. I could test the changes if someone handle this.

Gloomyeye, 2024-09-10:
@pimlie: I notice that you are active with fixing some bugs. Maybe you could help me with that bug. I would appreciate it.

MichielSchipper, 2025-02-09:
As a quick fix, I commented the line in qr.cpp where the password gets rendered:
//h = centerTextX(QR_WIFI_PASSWORD, 100, x - 100, y + (h + 30) * 2);

Gloomyeye, 2025-02-12:
Yes, thanks. I tested your workaround. The password is not displayed and therefore the qr code is now accessable. This worked for me.

lanrat, 2025-07-03:
I just pushed an update that will resize the WiFi information to fix in the available space based on the text length.

commit a856d7bd (2025-07-03): fixes #48