lanrat/homeplate #56: sleep_duration.h missing?

airmaxx, 2024-09-29:
I can't run pio. Is sleep_duration.h missing? I can see it should be included, but where the source?

```
Linking .pio/build/inkplate10v2/firmware.elf
/Users/airmax/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/inkplate10v2/src/activity.cpp.o:(.literal._Z13runActivitiesPv+0x20): undefined reference to `sleepScheduleSize'
/Users/airmax/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/inkplate10v2/src/activity.cpp.o:(.literal._Z13runActivitiesPv+0x24): undefined reference to `sleepSchedule'
collect2: error: ld returned 1 exit status
*** [.pio/build/inkplate10v2/firmware.elf] Error 1
```

lanrat, 2024-09-30:
@pimlie looks like your latest PRs forgot about the `#include "sleep_duration.h"` in `homeplate.h` and `test_get_sleep_duration.cpp`.

Do you mind updating them to fix this?

airmaxx, 2024-09-30:
Ah, after a few tests I realised that I had overlooked deleting this line from `config.cpp:`
`/* REMOVE THIS LINE TO ENABLE THE SLEEP SCHEDULE`

Now it also compiles when I want to use the custom sleep schedule.

Maybe the ticket will help someone with the same problem later.

lanrat, 2024-10-07:
Closing as this seems to be resolved. Feel free to comment or reopen if needed.

closed 2024-10-07 by lanrat