lanrat/homeplate #61: InkPlate6: crashing when trying to sleep?

ErwinP, 2025-07-05:
Hi,
First of all, thank you for this wonderful project. I've specifically bought and InkPlate for this, and was already planning for quite some time to play around with this.

I've purchased the InkPlate6 because of size restraints and managed to get it working - though I'm absolutely new to this, so forgive me if it's dirty! You can find my fork with the commit [here](https://github.com/ErwinP/homeplate). I'm at the stage where it boots, shows my homeassistant dashboard with proper width - I thought I was there! However, it doesn't update itself without being attached to my computer and when attached, I get frequent reboots. The log indicates something is going wrong when it goes into deep sleep, the log file is attached below.

I realize that you don't have an InkPlate6 and that you don't support it, but I was hoping for some advice nonetheless, or some pointers as to where the problem could be.

Thank you kindly in advance,
Erwin

```
21:10:23.193 > [SETUP] starting, version(Jul  5 2025, 21:09:45) boot: 0
21:10:23.193 > ESP Chip information:
21:10:23.193 > This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 1, 4MB external flash
21:10:23.193 > Minimum free heap size: 4416863 bytes
21:10:23.193 > Heap summary for capabilities 0x00000C06:
21:10:23.193 >   Totals:
21:10:23.193 >     free 0 allocated 0 min_free 0 largest_free_block 0
21:10:23.193 > [ACTIVITY] startActivity(1) put into queue
21:10:23.900 > [  1572][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:10:23.912 > [STATUS] Boot 1 normal
21:10:25.371 > [SETUP] Battery: 100% (4.22v)
21:10:25.749 > [SETUP] starting button task
21:10:25.749 > [INPUT] Input monitoring not supported by this platform
21:10:25.754 > [SETUP] starting time task
21:10:26.750 > [  4421][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:10:26.756 > [TIME] local time (3) 00:00:04 Thursday, January 01 1970
21:10:26.758 > [SETUP] starting WiFi task
21:10:26.761 > [WIFI] Connecting...
21:10:26.764 > [SETUP] starting OTA task
21:10:26.767 > [SETUP] starting MQTT task
21:10:26.769 > [MQTT] starting MQTT
21:10:26.769 > [SETUP] starting sleep task
21:10:26.772 > [SETUP] starting activities task
21:10:26.775 > [ACTIVITY] starting activity: 1
21:10:26.896 > [WIFI] Connected to AP successfully!
21:10:27.415 > [WIFI] IP address: 192.168.1.218
21:10:27.420 > [STATUS] WiFi connected
21:10:27.456 > [WIFI] Connected: 192.168.1.218
21:10:27.507 > [TIME] Syncing RTC to NTP
21:10:27.522 > [SLEEP] delaying sleep for 20 seconds
21:10:27.524 > [STATUS] Downloading image...
21:10:27.527 > [MQTT] Connecting to MQTT...
21:10:27.531 > [OTA] OTA ready
21:10:27.659 > [MQTT] Connected to MQTT.
21:10:27.659 > [MQTT] Session present: 0
21:10:27.662 > [MQTT] Subscribing at QoS 2, packetId: 1
21:10:27.668 > [MQTT] Sending MQTT Config
21:10:27.668 > [MQTT] Subscribe acknowledged: packetId: 1 qos: 2 
21:10:27.725 > [MQTT] Connected
21:10:28.899 > [IMAGE] Downloading image: http://<my-puppet-ip>:<port>/dashboard-homeplate/0?viewport=800x600&eink=2[NET] downloading file at URL http://<my-puppet-ip:>:<port>/dashboard-homeplate/0?viewport=800x600&eink=2
21:10:28.916 > [  6584][E][WiFiClient.cpp:320] setSocketOption(): fail on -1, errno: 9, "Bad file number"
21:10:29.255 > [IMAGE] Download done
21:10:29.256 > [STATUS] Rendering image...
21:10:29.900 > [  7572][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:10:29.907 > [TIME] Internal clock was adjusted by 1751742624 seconds
21:10:29.910 > [TIME] Internal RTC was adjusted by -173998060 seconds
21:10:29.915 > [TIME] NTP UNIX time Epoch(1751742629)
21:10:29.918 > [TIME] Timezone offset: (0) 0 hours
21:10:29.923 > [TIME] synced local UNIX time Epoch(1751742629) 19:10:29 Saturday, July 05 2025 
21:10:30.694 > [IMAGE] Detected image as PNG
21:10:31.642 > [  9314][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:10:31.648 > [TIME] ERROR: Failed to set RTC!
21:10:32.232 > [IMAGE] Image render ready
21:10:32.232 > [IMAGE] displaying....
21:10:33.779 > [IMAGE] displaying done.
21:10:34.770 > [MQTT] sending status update
21:10:34.771 > [MQTT] Sending MQTT State: [homeassistant/sensor/ARDUINO_ESP32_DEV/boot/state] {"boot":1,"activity_count":1,"boot_reason":"normal","sleep_duration":300}
21:10:34.984 > [MQTT] Sending MQTT State: [homeassistant/sensor/ARDUINO_ESP32_DEV/wifi_signal/state] {"signal":-73}
21:10:35.004 > [MQTT] Sending MQTT State: [homeassistant/sensor/ARDUINO_ESP32_DEV/temperature/state] {"temperature":26}
21:10:35.014 > [MQTT] Sending MQTT State: [homeassistant/sensor/ARDUINO_ESP32_DEV/battery/state] {"voltage":4.02,"battery":77}
21:10:47.772 > [ACTIVITY] startActivity(0) put into queue
21:10:47.774 > [ACTIVITY] starting activity: 0
21:10:47.777 > [SLEEP] prepping for sleep
21:10:47.780 > [MQTT] Stopping and disconnecting...
21:10:47.784 > [MQTT] Disconnected from MQTT.
21:10:47.787 > [WIFI] Stopping and disconnecting...
21:10:47.790 > [WIFI] Disconnected from WiFi access point
21:10:47.793 > [WIFI] WiFi lost connection. Reason: 8
21:10:47.861 > [SLEEP] entering sleep for 300 seconds (5 min)
21:10:47.864 > 
21:10:47.864 > 
21:10:47.912 > ets Jun  8 2016 00:22:57
21:10:47.912 > 
21:10:47.912 > rst:0x5 (DEEPSLEEP_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
21:10:47.918 > configsip: 0, SPIWP:0xee
21:10:47.921 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:10:47.926 > mode:DIO, clock div:2
21:10:47.929 > load:0x3fff0030,len:1184
21:10:47.932 > load:0x40078000,len:13232
21:10:47.936 > load:0x40080400,len:3028
  #0  0x40080400 in _init at ??:?

21:10:48.009 > entry 0x400805e4
21:10:48.457 > 
21:10:48.457 > 
21:10:48.457 > [SETUP] starting, version(Jul  5 2025, 21:09:45) boot: 1
21:10:48.460 > [ACTIVITY] startActivity(1) put into queue
21:10:49.514 > [  1545][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
21:10:49.533 > [STATUS] Boot 2 wake button
21:10:50.264 > [SETUP] Battery: 98% (4.19v)
21:10:50.264 > [SETUP] starting button task
21:10:50.266 > [INPUT] Input monitoring not supported by this platform
21:10:50.272 > [SETUP] starting time task
```

lanrat, 2025-07-07:
Hello @ErwinP,

Can you check that the `WAKE_BUTTON` is correct for the Inkplate6. On the Inkplate10 it is GPIO36, but it may be different on the Inkplate6. https://github.com/lanrat/homeplate/blob/main/src/homeplate.h#L36

Also verify if the wake button is active high or low. On the Inkplate10, the input goes low on wake. It could be different on the Inkplate6. https://github.com/lanrat/homeplate/blob/main/src/sleep.cpp#L56

closed 2025-07-09 by ErwinP

lanrat, 2025-07-09:
I'm glad to hear you found a solution.

I also just pushed an [update](https://github.com/lanrat/homeplate/commit/9b28011b268a4387481182f8519c2f3c4ed91d88) that should fix the issue for others as well.